PSP-Archive/ARK-4 #651: All ISO's crash on startup (unconsistent)

acelamo, 2025-09-16:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.69 r200

### Plugins ( if applicable )

ULUS01234, ms0:/SEPLUGINS/example.prx

### Steps to reproduce

1 i got a used psp 2000
2 with "skip sony logos" off the console doesnt boot into xmb, stuck at wave screen
3 with  "skip sony logos" "All" the console boots normally
4 with deafult ark 4 settings, the console shuts down after launching no matter wich iso game or file
5 deleting "isocache.bin" 90% of the times lets iso's run. but ive had exceptions
6 reseting ark 4 settings also worked to let iso run, but after some time the issue re appeared
7 the process of trying to run a iso: (1. clicking the file 2. the memory card indicator at left side blinks=game launches OR it doesnt blink most of the cases, then both wifi and power indicators flash green quickly, then the console shuts down.
8 i tried a different memory card, updated from r168 to r200, both did not work
conclusion: ARK 4 doesnt even read the memory stick after running iso files (indicator does not blink), the issue is a reappearing thing that blocks me from casual use of the psp, different random actions fix it but not pernamently and unconsistently. sometimes after a cold boot everything works, sometimes nothing helps but i try and change the inferno cache from LRU to OFF and it works, then it doesnt work again.  

### USB Charge

On

### CPU Clock in Game

Overclock

### CPU Clock in XMB

Overclock

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

All

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

On

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-09-16:
Reset settings and uninstall all plugins.

acelamo, 2025-09-17:
is there a simpler way? i got the psp with ark 4 installed, i dont want to brick it

acelamo, 2025-09-17:
can you give me a step by step tutorial how to reinstall ark using DC? 

> Try reinstalling the firmware using DC-ARK.