PSP-Archive/ARK-4 #653: Umd data install recognized by ark-4

Lucindavltn, 2025-09-19:
### New Feature

Hello,
Is it possible to have a new tool which can make data install  umd working with ark-4?

For games like akb 1/149 renai sousenkyo, data install is needed, but it only works on ofw, not cfw. It prevents me from playing the game

JoseAaronLopezGarcia, 2025-09-20:
I've never heard of this being an issue. Try nploader and/or npdrm_free.

Lucindavltn, 2025-09-20:
I have them both. It works with my others downloaded games and umds

But i cant play this game unless i boot my psp on ofw. But it prevents me from installing the game and consequently, use the dlc

The error message that i get is that i have no free space on the memory stick. I tried with 2 memory sticks

Lucindavltn, 2025-09-20:
I always get this message if i install the game with ark-4 
メモリースティック™の空き容量が足りません。

データインストールには

1600MB以上の空き容量が必要です

Which i translated and it said
There is not enough free space on the Memory Stick™.

Data installation requires at least 1600MB of free space.

On ofw i don't have this issue.
I even installed the game with ofw on each memory stick
Copy-pasted the game files on either one that i wasn't using 
I even formatted them both
The game data only install itself on ofw, with the two memory sticks 

I have a lot of free space

I have the same message with the iso file of the game

JoseAaronLopezGarcia, 2025-09-20:
Ok I see. I know the cause.

Lucindavltn, 2025-09-20:
Really ? Is it resolvable please?

JoseAaronLopezGarcia, 2025-09-20:
> Really ? Is it resolvable please?

Sorry but it was too late at night and I was already in bed. Try this: https://github.com/PSP-Archive/ARK-4/releases/tag/v1758373815

JoseAaronLopezGarcia, 2025-09-20:
Try installing anti-DRM plugins: nploader_mod, npdrm_free_mod and cfwblocker.

Lucindavltn, 2025-09-20:
I have these 3 plug-ins installed 
And cfblock from here
https://www.gamebrew.org/wiki/CFW_Folder_Blocker_PSP

I am on firmware 6.61 and even installed a plug-in which supports 6.60 plug-ins 

The game still doesn't work

Lucindavltn, 2025-09-20:
I have removed this plug-in 
I see all of my plugins on seplugins folder on my psp,except cfwblocker 
And i configured that file

I only see it with my pc

JoseAaronLopezGarcia, 2025-09-20:
Delete and reinstall the data.

Lucindavltn, 2025-09-20:
I did it, and the game umd still doesn't detect the install data

Lucindavltn, 2025-09-20:
It works now
What does it means?
Do I need to do that each time?

JoseAaronLopezGarcia, 2025-09-20:
> It works now What does it means? Do I need to do that each time?

It means one of the plugins you have installed is causing the issue (holding start when booting a game temporarily disables all plugins until you exit the game).
Disable one by one until you find which one is the problematic plugin.

Lucindavltn, 2025-09-20:
I found it, it was nploader mod

JoseAaronLopezGarcia, 2025-09-20:
Good to know. Thanks.

JoseAaronLopezGarcia, 2025-09-20:
For now just disable it.

Lucindavltn, 2025-09-20:
OK, thank you for your help
If it gets a new release, should I reinstall it in the future?