PSP-Archive/ARK-4 #654: Crash in God of War - Ghost of Sparta on rev 195 and later

schmarky, 2025-09-22:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

rev200 - The Last of Ass 

### Homebrew/GAME (if applicable)

UCES01401

### Plugins ( if applicable )

homebrew, leda.prx, on
always, bsod.prx, off

### Steps to reproduce

1. Enter Sparta 
2. Fight the brother
3. Approach the gate 
4. It crashes right after the guards shouts "Open the gate". 

This was not an issue in rev180. My PSP2000 and PSP3000 with Ark rev180 do not crash. After updating said PSP 3000 to rev200 the crash did occur.

This issue was also reported on Discord by a different user on the 07/29/25. Here it occurred on a PSP2000.

I can provide save games regarding this issue:
1. right before the crash
2. After entering the Gate which breaks the save file (I suspect this is connected to this issue)
3. The next savespot after that. From here on I can resume playing on rev200

### USB Charge

Off

### CPU Clock in Game

Auto

### CPU Clock in XMB

Auto

### WPA2

On

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Old Plugin Support on ef0

Off

### Prevent hibernation deletion

Off

### Use Extra Memory

Auto

### Memory Stick Speedup

On

### Inferno Cache

LRU

### Skip Sony Logos

All

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Off

### Disable UMD Drive

Off

### Disable Analog Stick

Off

### Enable QA Flags

Off

### Custom settings

always, usbcharge, on
always, overclock, on
always, wpa2, on
always, launcher, off
always, highmem, off
always, mscache, on
always, infernocache:lru, on
always, disablepause, off
always, oldplugin, on
always, hibblock, on
always, skiplogos, on
always, hidepics, off
always, hidemac, on
always, hidedlc, on
always, noled, off
always, noumd, off
always, noanalog, off
always, qaflags, on

# The following games don't like Inferno Cache
# Luxor - The Wrath of Set (the other Luxor game works fine)
ULUS10201, infernocache, off
# Flat-Out Head On (both US and EU)
ULUS10328 ULES00968, infernocache, off

# Enable Extra RAM on GTA LCS and VCS for CheatDeviceRemastered
ULUS10041 ULUS10160 ULES00151 ULES00502, highmem, on

### Verification of form.

- [x] I agree that I properly filled out the items listed above.

schmarky, 2025-09-22:
I did some further testing and it seems that the issue was introduced in rev186

schmarky, 2025-11-08:
Prior to r186. I did not have the issue on r180 for example.

JoseAaronLopezGarcia, 2025-11-08:
Try to narrow it down to a specific rev.

schmarky, 2025-11-08:
I already did in my very first comment. 
"I did some further testing and it seems that the issue was introduced in rev186"

schmarky, 2025-11-08:
Yes, r185 does work. The problem was introduced in r186.

schmarky, 2025-11-24:
The update provided by @krazynez in the previous comment did fix the issue for me. Thanks for the fix and keep up the good work.